mesh-iit/ventilator-FI5 #67: Missing Drawings

pattacini, 2020-05-16:
@mariovillalon from Mexico (where they aim to reproduce our FI5 prototype) forwarded us the following request:

>I found that there are some missing drawings of the custom components (in the Manufacturer column). Most of them have the category of "Loom". I would like to clarify what this term means. I understand that "Loom" refers to a special requirement part(?). However, I think we need drawings and/or a technical sheet to know what exactly these parts are. 
>
>Also, I noticed that we have some drawing numbers that are not in the BOM (see the second column below). What category do these parts belong to?
>
>Find attached a list with the number of both cases.

MISSING DRAWINGS OR TECHNICAL CHECK SHEET | MISSING NUMBER IN BOM
-- | --
VX-209515 | VX-209628
VX-209522 | VX-209631_1
VX-209583 | VX-209632_1
VX-209584 | VX-209635_1
VX-209585 | VX-209637_1
VX-209590 | VX-209639_1
VX-209594 | VX-209640_1
VX-209651 |  
VX-209750 |  
VX-209751 |  
VX-209648 |  
VX-209649 |  
VX-209652 |  
VX-209653 |  
VX-209655 |  
VX-209656 |  
VX-209658 |  
VX-209659 |  
VX-209705 |  
VX-209737 |  
VX-209744 |  
VX-209752 |  

cc @maggia80 @Fabrizio69

maggia80, 2020-05-16:
MISSING DRAWINGS OR TECHNICAL CHECK SHEET | NOTE
-- | --
VX-209515 | it's a 3D printed part. The drawing is missing, we will add it, but meanwhile, you can create the STL file from the .stp that you find in the https://github.com/icub-tech-iit/ventilator-FI5/blob/mech/VX-209503.stp [image]
VX-209522 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209583 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209584 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209585 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209590 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209594 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209651 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209750 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209751 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209648 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209649 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209652 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209653 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209655 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209656 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209658 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209659 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209705 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209737 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209744 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209752 | cable check the harness drawing and BOM https://github.com/icub-tech-iit/ventilator-FI5/tree/ele/wiring
VX-209628 | this code is not present in VX-209503.stp 
VX-209631_1 | parts for Manifold pressure test (not required for the prototype)
VX-209632_1 | parts for Manifold pressure test (not required for the prototype)
VX-209635_1 | parts for Manifold pressure test (not required for the prototype)
VX-209637_1 | parts for Manifold pressure test (not required for the prototype)
VX-209637_1 | parts for Manifold pressure test (not required for the prototype)  
VX-209639_1 | parts for Manifold pressure test (not required for the prototype)
VX-209640_1 | parts for Manifold pressure test (not required for the prototype)

maggia80, 2020-05-16:
@MarioVillalon let me know if there is something unclear or if you need further info.

MarioVillalon, 2020-05-16:
Gracias Marco, probaremos y te comento.

Saludos

maggia80, 2020-05-18:
@MarioVillalon we uploaded the missing drawing and we added the STL files for 3D printed parts.

pattacini, 2020-05-18:
@MarioVillalon for your convenience, you can download the latest `mech` release from here: https://github.com/icub-tech-iit/ventilator-FI5/releases/tag/mech-v1.0.1.

MarioVillalon, 2020-05-18:
Great Ugo, Thanks.

Best regards

MarioVillalon, 2020-05-27:
Ciao Ugo, hi Marco

Is it possible to get de document “FI5-specs.pdf” in Italian?

pattacini, 2020-05-27:
Hi @MarioVillalon 

Unfortunately, we don't have it in Italian.

MarioVillalon, 2020-05-27:
Ok Ugo, thanks anyways

Regards

closed 2020-06-02 by pattacini